flathub/com.prusa3d.PrusaSlicer #1: plater not using all screen real estate

alpharesearch, 2020-04-27:
As a test, I wanted to see how the flatpack works from the software manager in my Linux Mint 19.3 Mate 64bit. Looks like there is a bug in this version, the plater not using all screen real estate (is only 1/4 and on the bottom left, the rest of the area is app background). I'm using a 4K UHD resolution, maybe this is the issue. Btw, I'm using the PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage without this issue.

xarbit, 2020-04-27:
Thanks, can you please share a screenshot? Ill look in to this.

xarbit, 2020-04-27:
@alpharesearch 
Ok, I see what you mean. 
I tested this on my iMac 5k. It does work fine using the native resolution but not when I scaled it using the Display settings (200%).

xarbit, 2020-04-27:
interesting, using the RPM from the Fedora repos has the same issue..

xarbit, 2020-04-27:
It turns out it is a bug upstream when PrusaSlicer is built using wxGtk3. 
I opened a new issue: 
https://github.com/prusa3d/PrusaSlicer/issues/4135

I committed a new fix and changed to wxGtk2, that is also what the AppImage and there seems to be more issues open regarding wxGtk3 ..

closed 2020-04-27 by xarbit

alpharesearch, 2020-04-28:
Thanks for figuring this out.