flathub/com.prusa3d.PrusaSlicer #3: Locale issues

grigorig, 2020-04-29:
Thanks for the Flatpak of PrusaSlicer! It doesn't like my locale for some reason. I'm using Fedora 31 with a German UTF-8 locale and I get the following error:

```
$ flatpak run com.prusa3d.PrusaSlicer
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

It starts up correctly after unsetting `LC_*` environment variables. You should be able to reproduce this problem by setting the locale with `export LC_NUMERIC=de_DE.UTF-8`.

grigorig, 2020-04-29:
I'm not very familiar with Flatpak packaging, but this issue looks similar:

https://github.com/flatpak/flatpak/issues/1751

Apparently you need to explicitly enable some sort of locale extension to make system locales available in the container.

barthalion, 2020-04-29:
It's enabled automatically. Flatpak should have installed com.prusa3d.PrusaSlicer.Locale for you when you installed the app.

grigorig, 2020-04-29:
Yes, Flatpak installed that. Just reinstalled to validate that. No change!

xarbit, 2020-04-29:
Thanks, will check if I can reproduce this.

xarbit, 2020-04-29:
@grigorig I am using Fedora 32 and did the following. 

`export LC_NUMERIC=de_DE.UTF-8`
`flatpak run com.prusa3d.PrusaSlicer`

it works for me.. that ticket mentioned missing Platform.Locale.. 
While I have never encountered that issue, did you try : 

`flatpak install org.freedesktop.Platform.Locale/x86_64/19.08`

my result:
```
[xarbit@nuc-fritz-box ~]$ flatpak install org.freedesktop.Platform.Locale/x86_64/19.08
Looking for matches…
Remotes found with refs similar to ‘org.freedesktop.Platform.Locale/x86_64/19.08’:

   1) ‘flathub’ (system)
   2) ‘flathub’ (user)

Which do you want to use (0 to abort)? [0-2]: 1
Skipping: org.freedesktop.Platform.Locale/x86_64/19.08 is already installed
```

grigorig, 2020-04-29:
`org.freedesktop.Platform.Locale/x86_64/19.08` was installed on my system as well. This turned out to be some weird locale issue. Resetting the locale to en_US and then setting it to de_DE again made it work. I am not sure what happened behind the scenes.

In addition, the flatpak also Just Worked without any extra work required on another Fedora 31 system with similar locale setup. So this was some iffy configuration issue. It's weird that the PrusaSlicer flatpak seems to be literally the only application to have issues with it.

closed 2020-04-29 by grigorig

LatekVo, 2023-01-03:
Hi, i know this issue has been closed a long time ago but i just encountered it yet again, and nothing seems to fix it.
I'm using Arch Linux, and i downloaded Prusa Slicer from the AUR.
I've had it working previously this way, it's only now that it doesn't launch at all.
The errors are identical to the opening issue in this thread.
I'm tested this while using both the UTF-8 en_US, and the UTF-8 pl_PL locale.
I tried clearing LC_ALL, but that couldn't do it as well.

barthalion, 2023-01-03:
If you're using it from AUR, you should be discussing this on AUR.

xarbit, 2023-01-04:
@LatekVo we created a workaround using an entrypoint script to detect this rare but valid scenario. This was only done on PrusaSlicer for FlatHub. 
Have a look at what we have done below to work around this and please discuss this with the maintainer of the AUR package. 

See:
https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/master/entrypoint

LatekVo, 2023-01-05:
Thank you, will look into that.