flathub/com.prusa3d.PrusaSlicer #11: Gap around window

GammaGames, 2021-05-21:
When I run the app, the window has a gap and double titles:
[image]

xarbit, 2021-05-23:
haven't seen this, what type of screen are you using?

GammaGames, 2021-05-24:
Just a regular screen, 1080p with no scaling. I wonder if it's a GTK issue? I uninstalled `org.gtk.Gtk3theme.elementary` and the gap was smaller:

[image]

xarbit, 2021-09-02:
@GammaGames 
Can you please install the latest beta 2.4 and let me know if your issue still remains:

Install from command line as :

`flatpak install https://dl.flathub.org/build-repo/56101/com.prusa3d.PrusaSlicer.flatpakref
`

GammaGames, 2021-09-02:
All seems fixed! 
[image]
Love the clippy, too

xarbit, 2021-09-02:
awesome, thanks for testing!

closed 2021-09-02 by xarbit