flathub/com.prusa3d.PrusaSlicer #27: Locale issues with Norwegian locale, possibly others

einarjh, 2021-10-05:
Prusa Slicer won't launch when Norwegian measurement locales are set:

```
einar@prefect:~$ locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=nb_NO.UTF-8
LC_TIME=nb_NO.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=nb_NO.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=nb_NO.UTF-8
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=nb_NO.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

einar@prefect:~$ flatpak -v run com.prusa3d.PrusaSlicer 
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/einar/.local/share/flatpak
F: Opening user flatpak installation at path /home/einar/.local/share/flatpak
F: Opening user flatpak installation at path /home/einar/.local/share/flatpak
F: Cleaning up unused container id 2154689488
F: Cleaning up unused container id 4001615994
F: Allocated instance id 3847085963
F: Add defaults in dir /com/prusa3d/PrusaSlicer/
F: Add locks in dir /com/prusa3d/PrusaSlicer/
F: Allowing homedir access
F: Allowing x11 access
F: Running 'bwrap --args 40 xdg-dbus-proxy --args=42'
F: Running 'bwrap --args 40 prusa-slicer'
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

**Steps to reproduce:**

Using Fedora Workstation 34 with GNOME Shell, set the environment to:
* Language: English (United States)
* Formats: Norge (Norsk, bokmål) (or probably any non-English locale)

I saw in a previous [bug report](https://github.com/flathub/com.prusa3d.PrusaSlicer/issues/3) that the issue was resolved by setting all locales to US English and then back again to the desired locale (in that ticket it was de_DE), but this does not work for nb_NO (and doesn't seem to work for de_DE either).

com.prusa3d.PrusaSlicer.Locale is already installed:
```
einar@prefect:~$ flatpak list -a | grep prusa
PrusaSlicer	com.prusa3d.PrusaSlicer	2.3.3-1	stable	user
PrusaSlicer translations	com.prusa3d.PrusaSlicer.Locale		stable	user
```

Running the flatpak with LC_ALL=en_US.UTF-8 works, but this is a hassle, and is not something a regular user should have to deal with.

xarbit, 2021-10-07:
@einarjh  can you test it like this and report back:

`flatpak run --env=LC_ADDRESS=C --env=LC_COLLATE=C --env=LC_MONETARY=C --env=LC_MEASUREMENT=C --env=LC_NAME=C --env=LC_NUMERIC=C --env=LC_TELEPHONE=C com.prusa3d.PrusaSlicer`

einarjh, 2021-10-07:
Tested now, with it installed as system, same error message as before.

BUT. On a whim, I tried with only `--env=LANG=C`, and then it started right up. Also starts perfectly with LANG=en_US.UTF-8 and other `en_` variants.

xarbit, 2021-10-07:
@einarjh 

thanks, yes I know that works .. but it disables auto detection for languages that are supported by PrusaSlicer. 

can we try one more thing? 

`flatpak run --env=LC_CTYPE=C --env=LC_NUMERIC=C --env=LC_TIME=C --env=LC_COLLATE=C --env=LC_MONETARY=C --env=LC_MESSAGES=C --env=LC_PAPER=C --env=LC_NAME=C --env=LC_ADDRESS=C --env=LC_TELEPHONE=C --env=LC_MEASUREMENT=C --env=LC_IDENTIFICATION=C`

this will set all to C except the language.. 
Let me know if that works

xarbit, 2021-10-07:
this is what I found:

`On GNU systems, LC_ALL=C and LC_ALL=POSIX (or LC_MESSAGES=C|POSIX) override $LANGUAGE, while LC_ALL=anything-else wouldn't.`

xarbit, 2021-10-07:
@einarjh yes, it should be available soon

xarbit, 2021-10-07:
I have to revert the workaround ... it won't startup without using the terminal

xarbit, 2021-10-16:
@einarjh  This worked for me on Fedora 35 : 

Run: 

`flatpak update --appstream && flatpak update`

tested changing the locale and formats .. The above fixed it. 

I don't know why this is happening ..

peterge1998, 2021-11-04:
```
➜  ~ flatpak run com.prusa3d.PrusaSlicer  
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```
The following error happens when i try to run it on fedorra 35 too...

xarbit, 2021-11-04:
> ```
> ➜  ~ flatpak run com.prusa3d.PrusaSlicer
> An error occured while setting up locale.
> You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure [...]

Please run `flatpak update --appstream && flatpak update`

peterge1998, 2021-11-05:
> Please run `flatpak update --appstream && flatpak update`
> 
> What locale are you using?

```
➜  ~ flatpak update --appstream && flatpak update
Updating appstream data for remote fedora
Updating appstream data for remote flathub
Looking for updates…
Info: org.freedesktop.Platform.GL.default//19.08 is end-of-life, with reason:
   The Freedesktop SDK 19.08 runtime is no longer supported as of September 1, 2021. Please ask your application developer to migrate to a supported version
Nothing to do.
➜  ~ cat /etc/locale.conf 
LANG="en_US.UTF-8"
```

xarbit, 2021-11-05:
> > Please run `flatpak update --appstream && flatpak update`
> > What locale are you using?
> 
> ```
> ➜  ~ flatpak update --appstream && flatpak update
> Updating appstream data for remote fedora
> Updating appstream data for remote flathub
> Looking for updates…
> Info: org.freedesktop.Platform.GL.default//19.08 is end-of-life, with reason:
>    The Freedesktop SDK 19.08 runtime is no longer supported as of September 1, 2021. Please ask your application developer to migrate to a supported version
> Nothing to do.
> ➜  ~ cat /etc/locale.conf 
> LANG="en_US.UTF-8"
> ```
> 
> it is not starting after `flatpak update --appstream && flatpak update`...

I need more information, please post the output of `locale` command. What's interesting for me is do you have mixed locales? 
If that is the case, it usually works reinstalling if the some format locale was changed after the install. 

You mind trying the beta (upcoming 2.4) ?

`flatpak remote-add flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo`

xarbit, 2021-11-05:
I might force en_US.UTF-8 locale using env variable in the flatpak container until upstream (Prusa3d) has a fix, please also report you're issue here in this ticket: https://github.com/prusa3d/PrusaSlicer/issues/2365

peterge1998, 2021-11-05:
> I need more information, please post the output of `locale` command. What's interesting for me is do you have mixed locales? If that is the case, it usually works reinstalling if the some format locale was changed after the install.

```
➜  ~ locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=de_DE.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=de_DE.UTF-8
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT=de_DE.UTF-8
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

xarbit, 2021-11-05:
that's what I thought .. if you run `flatpak run --command=bash com.prusa3d.PrusaSlicer` and inside the container: `locale -a` what do you get?

peterge1998, 2021-11-05:
> You mind trying the beta (upcoming 2.4) ?
> 
> `flatpak remote-add flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo`

Gives the same error:
```
➜  ~ flatpak remote-list                            
Name         Options
fedora       system,oci
flathub      system
flathub-beta system
➜  ~ flatpak run com.prusa3d.PrusaSlicer/x86_64/beta
[2021-11-05 13:04:00.635803] [0x00007f11d57209c0] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```
---

> that's what I thought .. if you run `flatpak run --command=bash com.prusa3d.PrusaSlicer` then inside the container: `locale -a` what do you get?
```
➜  ~ flatpak run --command=bash com.prusa3d.PrusaSlicer
[📦 com.prusa3d.PrusaSlicer ~]$ locale -a
C
C.utf8
en_AG
en_AU
en_AU.utf8
en_BW
en_BW.utf8
en_CA
en_CA.utf8
en_DK
en_DK.utf8
en_GB
en_GB.utf8
en_HK
en_HK.utf8
en_IE
en_IE@euro
en_IE.utf8
en_IL
en_IN
en_NG
en_NZ
en_NZ.utf8
en_PH
en_PH.utf8
en_SC.utf8
en_SG
en_SG.utf8
en_US
en_US.utf8
en_ZA
en_ZA.utf8
en_ZM
en_ZW
en_ZW.utf8
POSIX
```

xarbit, 2021-11-06:
> ```
> ➜  ~ flatpak run --command=bash com.prusa3d.PrusaSlicer
> [📦 com.prusa3d.PrusaSlicer ~]$ locale -a
> C
> C.utf8
> en_AG
> en_AU
> en_AU.utf8
> en_BW
> en_BW.utf8
> en_CA
> en_CA.utf8
> en_DK
> en_DK.utf8
> en_GB [...]

as you can see, the reason why PrusaSlicer won't start is because the German locale is missing in the container. It seems to be passed by the OS but its missing in the container. 
Why this doesn't get resolved by reinstalling or executing `flatpak update --appstream && flatpak update` .. no clue yet, it works for me .. 
As a note, I was only able to reproduce this issue when installing PrusaSlicer first and changing the format locale after the install by something else using gnome settings ..  

Anyway,  I believe the locale detection code in the startup for PrusaSlicer needs to be adjusted on mixed locale and allow the app to start in any case it won't show anything after executing the desktop file and the user knows nothing. 

Maybe as a plus show a the configuration screen to configure the app language. (that screen does exist)
I also looked in the code myself and I am still trying to figure it out. 

As a workaround there seems to be two things you can do, which also helped others:

- https://github.com/flatpak/flatpak/issues/3712#issuecomment-918399466
- https://github.com/flatpak/flatpak/issues/3712#issuecomment-661308002

Prusa3D and myself need more input and help on solving this mystery. 

FYI: @bubnikv @lukasmatena

xarbit, 2021-11-06:
@einarjh @peterge1998 I am poking in the dark here a bit, would be great if you could help me here. 
I started a test build here : 

https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/31

maybe when done you could help and try it out with me .. ?

peterge1998, 2021-11-06:
> maybe when done you could help and try it out with me .. ?

I ran:
`➜  ~ flatpak install --user https://dl.flathub.org/build-repo/64093/com.prusa3d.PrusaSlicer.flatpakref`
and:
```
➜  ~ flatpak list | grep Prusa                                                                        
PrusaSlicer	com.prusa3d.PrusaSlicer	2.4.0-beta1	test	prusaslicer-origin	user
```
but still:
```
➜  ~ flatpak run com.prusa3d.PrusaSlicer
[2021-11-06 11:03:36.155456] [0x00007f51d72f19c0] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

xarbit, 2021-11-06:
You need to run as so: 

flatpak run com.prusa3d.PrusaSlicer//test

peterge1998, 2021-11-06:
> You need to run as so:
> 
> flatpak run com.prusa3d.PrusaSlicer//test
> 
> Update:
> 
> Oh I see you only have that one.

➜  ~ flatpak run com.prusa3d.PrusaSlicer//test
[2021-11-06 13:06:57.455948] [0x00007f1dc372b9c0] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid

peterge1998, 2021-11-19:
@xarbit Any news when 2.4.0 will be pushed live on flathub?

xarbit, 2021-11-20:
@peterge1998 

2.4.0 beta is already on flathub, as soon as the alphas started I pushed them to flathuib. 
If you want to give PrusaSlicer 2.4.0 Beta a shot you can do that by just enabling the flathub beta channel. 

I will push 2.4.0 to the release channel on flathub as soon as Prusa releases the software as stable. As you can see, we are at beta 1:

https://github.com/prusa3d/PrusaSlicer/releases

xarbit, 2021-11-28:
@peterge1998 @einarjh @bubnikv 

I created a little workaround for this issue with the mixed locale. It will be for the upcoming 2.4.x released flatpak. Ill be merging the code soon but I wanted to know if any of you can help out and test it before? Its considered a workaround and its a simple entrypoint script that basically detects a crash, reads the info and then it will just force LC_ALL=en_US.UTF-8 and boots up without notice. While I was on this  I also implemented a little bit of Dark Mode support :-) .. Please check it out and let me know if it works good enough or if it needs more attention. The code is here: https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/33 

Thanks !

peterge1998, 2021-11-28:
> @peterge1998 @einarjh @bubnikv
>
> I created a little workaround for this issue with the mixed locale. It will be for the upcoming 2.4.x released flatpak. Ill be merging the code soon but I wanted to [...]

```
➜  ~ flatpak install --user https://dl.flathub.org/build-repo/66610/com.prusa3d.PrusaSlicer.flatpakref

com.prusa3d.PrusaSlicer permissions:
    ipc                     network                 x11                       devices                        dri
    file access [1]         dbus access [2]         bus ownership [3]         system dbus access [4]

    [1] /media, /run/media, home
    [2] com.prusa3d.prusaslicer.InstanceCheck.*, org.freedesktop.Flatpak
    [3] com.prusa3d.prusaslicer.*
    [4] org.freedesktop.UDisks2

        ID                                         Branch            Op            Remote                        Download
 1. [✓] com.prusa3d.PrusaSlicer.Locale             test              i             prusaslicer-origin            220,1 kB / 3,4 MB
 2. [✓] com.prusa3d.PrusaSlicer                    test              i             prusaslicer-origin             44,1 MB / 45,6 MB

Installation complete.
➜  ~ flatpak run com.prusa3d.PrusaSlicer       

[2021-11-28 21:52:24.327581] [0x00007f7ffeff79c0] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

xarbit, 2021-11-29:
@peterge1998 .. thanks for testing. As a note this workaround is currently only activated when using the desktop files. running from Command-line will not work unless you run it as the following :-) :

`flatpak run --command=entrypoint com.prusa3d.PrusaSlicer`

peterge1998, 2021-11-30:
> `flatpak run --command=entrypoint com.prusa3d.PrusaSlicer`

Sry, did not know this. Anyway, 2.4.0 runs fine on my setup :) Thanks for adressing this problem!

xarbit, 2021-12-05:
fixed with workaround for 2.4.0, closing

closed 2021-12-05 by xarbit

xarbit, 2021-12-05:
https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/33 is now merged in the beta channel/branch on flathub and will work when running  `flatpak run com.prusa3d.PrusaSlicer`, I mean without using `--command=entrypoint`. 

you can get the official flathub beta channel by installing this repo: 

`
flatpak remote-add flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
`

would be great to know  if it works for you. 
The dark variant theme  looks great :-) 

[image]

Bricksone, 2022-08-11:
It's not working for me (version 2.4.2 from flathub):
"[xyz@xyz ~]$ flatpak run com.prusa3d.PrusaSlicer
--------------------------------------------------------------------------
Message: 13:46:07: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
/app/bin/entrypoint: line 12: warning: setlocale: LC_ALL: cannot change locale (en_DE.UTF-8)
Message: 13:46:07: WARN: Executing prusa-slicer with locale workaround en_DE.UTF-8
[patson@patson-ms7b86 ~]$ [2022-08-11 13:46:07.632571] [0x00007fd7b1b519c0] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid"

xarbit, 2022-08-11:
@Bricksonecan you give me more information about your setup and what you are doing here? 
Trying to understand the [patson@patson-ms7b86 ~] session ..

Bricksone, 2022-08-11:
@xarbit  Thanks for the reply. The patson part should've been xyz as well as in the beginning. Just forgot to edit it. I am new to posting on github ¯\_(ツ)_/¯

I am running EndeavourOS (Plasma, Wayland) and installed Prusa Slicer as a flatpak. Then I used _flatpak run com.prusa3d.PrusaSlicer_ to run it. Everything below that is the output I got in the terminal. (Above and below _Message: 13:46:07: Starting PrusaSlicer flatpak with entrypoint script_ there were dotted lines that messed up the formatting. So I removed them)

@xarbit  The patson part should've been xyz as well as in the beginning. Just forgot to edit it. I am new to posting on github ¯\_(ツ)_/¯

I am running EndeavourOS (Plasma, Wayland) and installed Prusa Slicer as a flatpak. Then I used _flatpak run com.prusa3d.PrusaSlicer_ to run it. Everything below that is the output I got in the terminal. (Above and below _Message: 13:46:07: Starting PrusaSlicer flatpak with entrypoint script_ there were dotted lines that messed up the formatting. So I removed them)

Prusa Slicer was only working for me (arch repo, AUR or flathub) if my formats settings for region, numbers, time, currency and measurements were the same (e.g. en_US.UTF-8). But I wanted a mix between Englisch (system language) and German (rest). After some more research it looked like it's a Plasma problem that's only affecting some programs (FreeCAD and Blender for example made no problems).

Using locale-gen as the error message suggested said that it generated locales de_DE.UTF-8 and en_US.UTF-8. Just what I needed. But it still didn't work.
I then checked my /etc/locales.gen and saw that both locales were uncommented at the end (under "# Locales generated by Calamares"), but not further up in the list. So I uncommented both of them further up in the list as well and ran locale-gen again, which now generated both of them twice. Doesn't make any sense to me, but now Prusa Slicer is actually working.

commit 7ab14cdc (2022-03-11): fixes #27