flathub/com.prusa3d.PrusaSlicer #41: App does not start on Silverblue 35 with Wayland

shoop, 2022-01-15:
```
$ flatpak run -v com.prusa3d.PrusaSlicer
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/stijn/.local/share/flatpak
F: Opening user flatpak installation at path /var/home/stijn/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/stijn/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.freedesktop.Platform/x86_64/21.08/b27db466e56f7e49e794656b84153c8b81f929ca8b5b8400142b19bb48b9a63d/files/lib32 does not exist
F: Cleaning up unused container id 968333733
F: Cleaning up per-app-ID state for com.prusa3d.PrusaSlicer
F: Allocated instance id 3021925260
F: Add defaults in dir /com/prusa3d/PrusaSlicer/
F: Add locks in dir /com/prusa3d/PrusaSlicer/
F: Allowing homedir access
F: Allowing x11 access
F: Running 'bwrap --args 38 xdg-dbus-proxy --args=40'
F: Running 'bwrap --args 38 entrypoint'
--------------------------------------------------------------------------
Message: 10:57:37: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
/app/bin/entrypoint: line 16: [: ==: unary operator expected
stijn@tangaloor: ~
$ [2022-01-15 10:57:37.682435] [0x00007f2e35cd59c0] [trace]   Initializing StaticPrintConfigs
Gtk-Message: 10:57:37.717: Failed to load module "canberra-gtk-module"
Gtk-Message: 10:57:37.717: Failed to load module "canberra-gtk-module"

(prusa-slicer:8): Gdk-ERROR **: 10:57:38.168: The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAlloc (insufficient resources for operation)'.
  (Details: serial 2407 error_code 11 request_code 147 (unknown) minor_code 7)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
```

Not sure if the root cause is the missing "canberra-gtk-module" but I tried to force it to use Wayland but failed.

xarbit, 2022-01-17:
Wayland is not supported with the version of wxWidgets used by Prusa. 
So are you saying, the app does not work even though Wayland is not used in the sockets permission? 

For the time being PrusaSlicer is X11 only and Wayland is not referenced. 
https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/904b3f5b3e503d80b0eb2d8b025d81914df81b1a/com.prusa3d.PrusaSlicer.yml#L13

xarbit, 2022-01-17:
the missing "canberra-gtk-module" warning is very normal when using X11 .. 
can you set it back to X11 and just try it with : 

`flatpak override --env=PRUSA_SLICER_DARK_THEME=false com.prusa3d.PrusaSlicer`

or true if you like.

shoop, 2022-01-17:
hi @xarbit , I think this is not specific to this Flatpak actually. The error I saw in #42 did appear in the log but the crash was later. I believe that it is the upstream issue flatpak/flatpak#4629 that is tripping me up, so feel free to close this one in favor of the upstream one.

closed 2022-01-17 by xarbit