flathub/com.prusa3d.PrusaSlicer #44: Unable to unmount SD card

xvalen, 2022-02-10:
**Version**
2.4.0

**Operating system type + version**
OpenSUSE Tumbleweed 20220207
Fedora 35

**Issue:**
I am unable to unmout SD card or USB stick after exporting gcode. Application shows alert that Ejecting failed. I have not found nothing useful in logs and internet. When I install prusaslicer from package manager, unmounting works. 

What could be wrong please?

xarbit, 2022-04-26:
Does this issue still remain?

jhaand, 2022-06-08:
Yes, the issue still persists with this flatpak.

Tested on Debiab Bookworm /  KDE 5.90.0 / Plasma 5.24.4

eliadevito, 2022-09-28:
Ubuntu snap version seem to apply a patch for this: https://github.com/ivocavalcante/PrusaSlicer-snap/blob/main/patches/use_udisks2_for_unmounting.patch maybe it can be helpful here or directly upstream.

@xarbit what do you think about it?

xarbit, 2022-10-14:
@eliadevito  As you already seen, I added it in .. Can you test it and give feedback so we can close this ticket? Thanks

bubnikv, 2022-10-14:
> Ubuntu snap version seem to apply a patch for this:
https://github.com/ivocavalcante/PrusaSlicer-snap/blob/main/patches/use_udisks2_for_unmounting.patch
maybe
it can be helpful here or directly upstream.

Thanks, we will look at that patch.

xarbit, 2022-10-19:
Fixed in https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/54 .. @eliadevito thanks for the good work

closed 2022-10-19 by xarbit

jhaand, 2022-10-20:
I just installed PrusaSlicer 2.5.0. Unmounting the USB pen drive now works from PrusaSlicer.