flathub/com.prusa3d.PrusaSlicer #46: Update to 2.5.0

kozross, 2022-09-10:
This update's been available since September 7th, and it would be nice to have its new cool features.

dsm, 2022-09-12:
any progress ?

leon-matthews, 2022-09-14:
I'm really looking forward to this update. The Arachne engine looks fantastic. Thank you!

bear454, 2022-09-16:
https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/48

xarbit, 2022-09-21:
- https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/49
- https://github.com/prusa3d/PrusaSlicer/issues/8922
- https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/48

xarbit, 2022-09-22:
@bear454 @leon-matthews @dsm @kozross 

we have build that works, there was a problem with TBB. 
But before I can merge this to master, can you please test the test build and let me know if it works for you as expected? 

`flatpak install https://dl.flathub.org/build-repo/109209/com.prusa3d.PrusaSlicer.flatpakref`

xarbit, 2022-09-22:
Thanks @Teeffelen .. Upstream is notified and lets see if we can get this fixed.

bear454, 2022-09-22:
Should this block the release? AFAIK this is happening in the upstream 2.5.0 as well.

xarbit, 2022-09-22:
I will not release it to flathub if broken that is for sure. I'd rather wait for 2.5.1. 
But let's wait and see what upstream finds and hope for a hotfix that works.

bear454, 2022-09-22:
Ack. For reference, I've done ~30 slices for Mk3S+ & MMU2S ... no errors/crashes with the 2.5.0 AppImage. I'll just keep using that one in parallel to the 2.4.2 flathub when applicable.

leon-matthews, 2022-09-22:
> @bear454 @leon-matthews @dsm @kozross
> Can you please test the test build and let me know if it works for you as expected?

App loaded well, but crashed to desktop with the very first model I sliced:

`/usr/include/c++/12.1.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.`

bear454, 2022-09-23:
```
$ flatpak install --user https://dl.flathub.org/build-repo/109209/com.prusa3d.PrusaSlicer.flatpakref

com.prusa3d.PrusaSlicer permissions:
    ipc                            network                 x11                    devices
    dri                            file access [1]         dbus access [2]        bus ownership [3]
    system dbus access [4]

    [1] /media, /run/media, home
    [2] com.prusa3d.prusaslicer.InstanceCheck.*, org.freedesktop.DBus.Introspectable.*
    [3] com.prusa3d.prusaslicer.*
    [4] org.freedesktop.UDisks2

        ID                                 Branch     Op     Remote                 Download
 1. [✓] com.prusa3d.PrusaSlicer.Locale     test       i      prusaslicer-origin     108.1 kB / 9.2 MB
 2. [✓] com.prusa3d.PrusaSlicer            test       i      prusaslicer-origin      30.0 MB / 57.3 MB

Installation complete.
$ flatpak run com.prusa3d.PrusaSlicer 
--------------------------------------------------------------------------
Message: 17:04:29: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
[2022-09-22 17:04:29.755947] [0x00007f55050147c0] [trace]   Initializing StaticPrintConfigs
Gtk-Message: 17:04:29.764: Failed to load module "appmenu-gtk-module"
Gtk-Message: 17:04:29.835: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:04:29.835: Failed to load module "canberra-gtk-module"

(prusa-slicer:8): Gtk-CRITICAL **: 17:05:46.361: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:8): Gtk-CRITICAL **: 17:05:46.361: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:8): Gtk-CRITICAL **: 17:05:46.504: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(prusa-slicer:8): Gtk-CRITICAL **: 17:05:46.505: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
/usr/include/c++/12.1.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.
```

I got that crash on the first slice, using the same settings that succeeded with the AppImage (GTK3 version).

closed 2022-10-19 by xarbit