flathub/com.prusa3d.PrusaSlicer #51: Looking for new Maintainer

xarbit, 2022-10-10:
Unfortunately, I need to focus on other things and I just do not have the necessary time or the proper Linux hardware anymore to keep maintaining the prusa-slicer flatpak as I have fully switched to macOS only. 

If anyone wants to take over this, please do so and let me know in the comments below.
Otherwise I'll need to flag this package as unmaintained in the flathub repository in the coming days or so. 

FYI:
@tamasmeszaros , @bubnikv

Emporea, 2022-10-10:
What knowledge do I need to maintain something like this?

xarbit, 2022-10-11:
@Emporea 

what is your background? Maintaining this would require a decent foundation and affinity in software development and Linux.

Emporea, 2022-10-11:
I dont think i would be capable of doing this without a crash course.

eliadevito, 2022-10-12:
Hi xarbit, I'm available to maintain this, tell me if for you are ok.

xarbit, 2022-10-14:
I am absolutely ok with it, thanks for jumping in. :-)

closed 2022-10-14 by xarbit