flathub/com.prusa3d.PrusaSlicer #58: Can't open PrusaSlicer on the Steam Deck with non English locales.

DiThi, 2023-01-07:
The app icon does nothing, and when I run it in console it shows this:

```
(deck@steamdeck ~)$ com.prusa3d.PrusaSlicer 
--------------------------------------------------------------------------
Message: 19:00:51: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
/app/bin/entrypoint: line 12: warning: setlocale: LC_ALL: cannot change locale (es_ES.UTF-8)
Message: 19:00:51: WARN: Executing prusa-slicer with locale workaround es_ES.UTF-8
[2023-01-07 19:00:51.235063] [0x00007f0fffd55a00] [trace]   Initializing StaticPrintConfigs
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

Running locale-gen doesn't fix it, I think that has to be something run when the package is built.

Running it like this, works: `LANG=C com.prusa3d.PrusaSlicer`

This is the console output when it runs successfully (nothing special I think):

```
--------------------------------------------------------------------------
Message: 19:05:23: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
[2023-01-07 19:05:23.821068] [0x00007fcf3c274a00] [trace]   Initializing StaticPrintConfigs
Gtk-Message: 19:05:23.836: Failed to load module "canberra-gtk-module"
```

I don't have this problem with any of the other 76 flatpaks I have installed.

OS: SteamOS Holo x86_64 (beta channel)
DE: Plasma 5.26.1 
WM: KWin X11
Theme: Vapor [Plasma], Breeze [GTK2/3]

xarbit, 2023-04-05:
we cannot support steam deck as we do not have the hardware

bubnikv, 2023-04-05:
You may consider to do what the message suggests:

> You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.

xarbit, 2024-09-28:
I found the issue ..

xarbit, 2024-09-28:
fix: https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/124

xarbit, 2024-09-28:
Build [150301 successful](https://buildbot.flathub.org/#/builders/6/builds/150301)
To test this build, install it from the testing repository:
```
flatpak install --user https://dl.flathub.org/build-repo/133392/com.prusa3d.PrusaSlicer.flatpakref
```

PR #124 (2024-09-28): fix: #121 + #58
fix: #58 and #121
closed 2024-09-28 by xarbit