flathub/com.prusa3d.PrusaSlicer #67: not inheriting system color scheme

kinghat, 2023-03-31:
i have my system theme in dark mode but it doesnt carry to prusaslicer. 

Operating System: Fedora Linux 37
KDE Plasma Version: 5.27.3
KDE Frameworks Version: 5.104.0
Qt Version: 5.15.8
Kernel Version: 6.2.8-200.fc37.x86_64 (64-bit)
Graphics Platform: Wayland

this is the env var set from flatseal:

[image]

eliadevito, 2023-04-02:
Currently PrusaSlicer doesn't handle dark/light theme switch automatically.

To enable dark theme you have to set PRUSA_SLICER_DARK_THEME to true

kinghat, 2023-04-02:
> Currently PrusaSlicer doesn't handle dark/light theme switch automatically.
> 
> To enable dark theme you have to set PRUSA_SLICER_DARK_THEME to true

do you know how that is being set on other systems? when i asked in discord, they said theirs was set automatically. im guessing they were on win/mac.

xarbit, 2023-04-05:
in flatpak it will not get automatically detected, PrusaSlicer is not a native Gtk app. 
The workaround we created that uses an internal PRUSA_SLICER_DARK_THEME  flag is currently the only way to get it dark using FlatPak.

closed 2023-04-05 by xarbit