flathub/com.prusa3d.PrusaSlicer #80: don't use filesystem=home

alexmo1997, 2023-07-29:
Using filesystem=home is not really a good practice as it allows access to the whole user home directory which is not needed here.

xarbit, 2023-08-27:
That is correct, the reason we use it is because PrusaSlicer and wxWidgets do not support portals API yet. 
How else would we provide access to the users home directory so they can load and save their projects and stl's?

alexmo1997, 2023-08-27:
Well, correct me if I'm wrong, but from another project I seem to remember that wxwidgets now does in fact support xdg portals already.

xarbit, 2023-08-27:
I am not aware of this yet, would definitely be cool. 
We are also still waiting for proper Wayland support for wxWidgets. 

Prusa also has its own fork of wxWidgets and for what its worth, they would need to update their code to support it never the less. 

As soon as PrusaSlicer is there, we will certainly adopt it.

But for now I will close this issue.

closed 2023-08-27 by xarbit