flathub/com.prusa3d.PrusaSlicer #89: Cannot resolve physical printer hostname over multicast DNS (*.local)

Hyperspeed1313, 2023-10-20:
I can't get PrusaSlicer to find my printer by hostname. The printer uses Fluidd/Mainsail and broadcasts its hostname over multicast dns (fluiddpi.local). Adding an entry for the printer to my network's DNS rewrites allows it to be found (assuming the DNS rewrite doesn't use the .local TLD).

I've thrown every flatseal setting I can at this package, but I'm still unable to resolve the hostname via mDNS using the Flatpak package - testing the hostname gets [Error 6] - could not resolve host. This is not a problem on Windows or on the AppImage. I've tried throwing literally every setting Flatseal allows at it but none have helped.

eliadevito, 2023-10-20:
This should be caused by a flatpak limitation tracked here [1013](https://gitlab.com/freedesktop-sdk/freedesktop-sdk/-/issues/1013)

Hyperspeed1313, 2023-10-20:
Until that issue is closed and it's still a problem, concur.

closed 2023-10-20 by Hyperspeed1313