flathub/com.prusa3d.PrusaSlicer #99: Cannot choose language in PrusaSlicer 2.7.1

jernejp21, 2024-02-01:
I think that flatpak build went wrong somewhere. When I install flatpak from flathub, I cannot choose language other than English.

[image]

I built my own flatpak and installed it and language selection works, but when selecting other language (Japanese for example), I get this error message.
Install command: `flatpak-builder --user --install --force-clean build-dir com.prusa3d.PrusaSlicer.yml`

[image]

Since flatpak is in sandbox, I think locales for all supported languages should also be build. But the question is still why is only English shown.

One more thing, in `/home/<user>/.local/share/flatpak/com.prusa3d.PrusaSlicer/current/active/files/share/runtime/locale`, all folders are empty. They should have po and mo files inside.

eliadevito, 2024-02-10:
Hi @jernejp21,
You have to enable the locale on flatpak, to do it you have to set the languages/extra-languages  key with  `flatpak config` as described here: [https://docs.flatpak.org/en/latest/flatpak-command-reference.html#flatpak-config](https://docs.flatpak.org/en/latest/flatpak-command-reference.html#flatpak-config).

If you need help to do it, please post here the output of `flatpak config --list` command and the locale that you want to enable

jernejp21, 2024-02-10:
@eliadevito 

Thank you for your help. Your proposal worked.

closed 2024-02-10 by jernejp21