flathub/com.prusa3d.PrusaSlicer #100: Cannot download anything via desktop integration

konradmb, 2024-02-03:
When opening links from printables.com, Prusa Slicer opens up and shows the following dialog:
[image]
<details><summary>Console output:</summary>
<p>

```
xdg-open 'prusaslicer://open?file=https%3A%2F%2Ffiles.printables.com%2Fmedia%2Fprints%2F745432%2Fstls%2F5846564_369e98f7-f935-4096-b3a2-e30a0f98d3b8_1607d8bf-01ba-446b-ba18-42bdad01f08a%2Fclearence_test_02_hard.stl' 

--------------------------------------------------------------------------
Message: 20:36:34: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
Gtk-Message: 20:36:34.665: Failed to load module "xapp-gtk3-module"
Gtk-Message: 20:36:34.694: Failed to load module "canberra-gtk-module"
Gtk-Message: 20:36:34.694: Failed to load module "pk-gtk-module"
Gtk-Message: 20:36:34.695: Failed to load module "canberra-gtk-module"
Gtk-Message: 20:36:34.696: Failed to load module "pk-gtk-module"

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.469: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.469: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.471: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.471: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.475: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.476: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.481: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.482: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.825: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.825: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.843: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.843: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.844: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.844: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.846: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.847: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.847: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.849: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.849: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.849: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.849: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.850: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.850: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.850: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.850: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.850: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:8): Gtk-CRITICAL **: 20:36:35.853: gtk_widget_set_size_request: assertion 'width >= -1' failed
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
20:36:36: Debug: window wxTreeCtrl@0x563080ee3450 ("treeCtrl") lost focus even though it didn't have it
[2024-02-03 20:36:37.757923] [0x00007f744a911b40] [error]   stl_open_count_facets: The input is an empty file: /home/konrad/Pobrane/STL/clearence_test_02_hard(2).stl
```

</p>
</details> 

The output is indeed an empty file.

There's a similar issue on Nix: https://github.com/NixOS/nixpkgs/issues/246261
I've tested official AppImage and it works.

eliadevito, 2024-02-09:
Hi @konradmb,
thanks for you report, I'm able to reproduce this, I will investigate for a fix in the next days.

@xarbit I keep you updated

eliadevito, 2024-02-15:
fixed for me by

konradmb, 2024-02-15:
@eliadevito Just tested it, works great! Thank you very much!

konradmb, 2024-02-18:
Closing. Stable version from flathub is working for me.

closed 2024-02-18 by konradmb