flathub/com.prusa3d.PrusaSlicer #121: Stops startup with"An error occured while setting up locale."

CharlemagneLasse, 2024-09-01:
The appimage is working perfectly fine but the flatpak application is refusing to start up

```
flatpak run com.prusa3d.PrusaSlicer                   
--------------------------------------------------------------------------
Message: 13:15:30: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
/app/bin/entrypoint: line 21: warning: setlocale: LC_ALL: cannot change locale (de_DE.UTF-8)
Message: 13:15:30: WARN: Executing prusa-slicer with locale workaround de_DE.UTF-8
An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
PrusaSlicer will now terminate.

locale::facet::_S_create_c_locale name not valid
```

It seems like the locales were not installed in the flatpak container:

```
$ flatpak run --command=bash com.prusa3d.PrusaSlicer                                                                                                
$ locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=de_DE.UTF-8
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC=en_US.UTF-8
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=
$ locale -a
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_COLLATE to default locale: No such file or directory
C
C.utf8
POSIX
en_AG
en_AU
en_AU.utf8
en_BW
en_BW.utf8
en_CA
en_CA.utf8
en_DK
en_DK.utf8
en_GB
en_GB.utf8
en_HK
en_HK.utf8
en_IE
en_IE.utf8
en_IE@euro
en_IL
en_IN
en_NG
en_NZ
en_NZ.utf8
en_PH
en_PH.utf8
en_SG
en_SG.utf8
en_US
en_US.utf8
en_ZA
en_ZA.utf8
en_ZM
en_ZW
en_ZW.utf8
```

And on my actual system, I actually see the correct locales:

```
$ locale -a
aa_DJ
aa_DJ.utf8
aa_ER
aa_ER@saaho
aa_ET
af_ZA
af_ZA.utf8
agr_PE
ak_GH
am_ET
an_ES
an_ES.utf8
anp_IN
ar_AE
ar_AE.utf8
ar_BH
ar_BH.utf8
ar_DZ
ar_DZ.utf8
ar_EG
ar_EG.utf8
ar_IN
ar_IQ
ar_IQ.utf8
ar_JO
ar_JO.utf8
ar_KW
ar_KW.utf8
ar_LB
ar_LB.utf8
ar_LY
ar_LY.utf8
ar_MA
ar_MA.utf8
ar_OM
ar_OM.utf8
ar_QA
ar_QA.utf8
ar_SA
ar_SA.utf8
ar_SD
ar_SD.utf8
ar_SS
ar_SY
ar_SY.utf8
ar_TN
ar_TN.utf8
ar_YE
ar_YE.utf8
as_IN
ast_ES
ast_ES.utf8
ayc_PE
az_AZ
az_IR
be_BY
be_BY@latin
be_BY.utf8
bem_ZM
ber_DZ
ber_MA
bg_BG
bg_BG.utf8
bhb_IN.utf8
bho_IN
bho_NP
bi_VU
bn_BD
bn_IN
bo_CN
bo_IN
br_FR
br_FR@euro
br_FR.utf8
brx_IN
bs_BA
bs_BA.utf8
byn_ER
C
ca_AD
ca_AD.utf8
ca_ES
ca_ES@euro
ca_ES.utf8
ca_ES@valencia
ca_FR
ca_FR.utf8
ca_IT
ca_IT.utf8
ce_RU
chr_US
ckb_IQ
cmn_TW
crh_UA
csb_PL
cs_CZ
cs_CZ.utf8
C.utf8
cv_RU
cy_GB
cy_GB.utf8
da_DK
da_DK.utf8
de_AT
de_AT@euro
de_AT.utf8
de_BE
de_BE@euro
de_BE.utf8
de_CH
de_CH.utf8
de_DE
de_DE@euro
de_DE.utf8
de_IT
de_IT.utf8
de_LI.utf8
de_LU
de_LU@euro
de_LU.utf8
doi_IN
dsb_DE
dv_MV
dz_BT
el_CY
el_CY.utf8
el_GR
el_GR@euro
el_GR.utf8
en_AG
en_AU
en_AU.utf8
en_BW
en_BW.utf8
en_CA
en_CA.utf8
en_DK
en_DK.iso885915
en_DK.utf8
en_GB
en_GB.iso885915
en_GB.utf8
en_HK
en_HK.utf8
en_IE
en_IE@euro
en_IE.utf8
en_IL
en_IN
en_NG
en_NZ
en_NZ.utf8
en_PH
en_PH.utf8
en_SC.utf8
en_SG
en_SG.utf8
en_US
en_US.iso885915
en_US.utf8
en_ZA
en_ZA.utf8
en_ZM
en_ZW
en_ZW.utf8
eo
es_AR
es_AR.utf8
es_BO
es_BO.utf8
es_CL
es_CL.utf8
es_CO
es_CO.utf8
es_CR
es_CR.utf8
es_CU
es_DO
es_DO.utf8
es_EC
es_EC.utf8
es_ES
es_ES@euro
es_ES.utf8
es_GT
es_GT.utf8
es_HN
es_HN.utf8
es_MX
es_MX.utf8
es_NI
es_NI.utf8
es_PA
es_PA.utf8
es_PE
es_PE.utf8
es_PR
es_PR.utf8
es_PY
es_PY.utf8
es_SV
es_SV.utf8
es_US
es_US.utf8
es_UY
es_UY.utf8
es_VE
es_VE.utf8
et_EE
et_EE.iso885915
et_EE.utf8
eu_ES
eu_ES@euro
eu_ES.utf8
eu_FR
eu_FR@euro
eu_FR.utf8
fa_IR
ff_SN
fi_FI
fi_FI@euro
fi_FI.utf8
fil_PH
fo_FO
fo_FO.utf8
fr_BE
fr_BE@euro
fr_BE.utf8
fr_CA
fr_CA.utf8
fr_CH
fr_CH.utf8
fr_FR
fr_FR@euro
fr_FR.utf8
fr_LU
fr_LU@euro
fr_LU.utf8
fur_IT
fy_DE
fy_NL
ga_IE
ga_IE@euro
ga_IE.utf8
gd_GB
gd_GB.utf8
gez_ER
gez_ER@abegede
gez_ET
gez_ET@abegede
gl_ES
gl_ES@euro
gl_ES.utf8
gu_IN
gv_GB
gv_GB.utf8
hak_TW
ha_NG
he_IL
he_IL.utf8
hif_FJ
hi_IN
hne_IN
hr_HR
hr_HR.utf8
hsb_DE
hsb_DE.utf8
ht_HT
hu_HU
hu_HU.utf8
hy_AM
hy_AM.armscii8
ia_FR
id_ID
id_ID.utf8
ig_NG
ik_CA
is_IS
is_IS.utf8
it_CH
it_CH.utf8
it_IT
it_IT@euro
it_IT.utf8
iu_CA
ja_JP.eucjp
ja_JP.utf8
kab_DZ
ka_GE
ka_GE.utf8
kk_KZ
kk_KZ.rk1048
kk_KZ.utf8
kl_GL
kl_GL.utf8
km_KH
kn_IN
kok_IN
ko_KR.euckr
ko_KR.utf8
ks_IN
ks_IN@devanagari
ku_TR
ku_TR.utf8
kw_GB
kw_GB.utf8
ky_KG
lb_LU
lg_UG
lg_UG.utf8
li_BE
lij_IT
li_NL
ln_CD
lo_LA
lt_LT
lt_LT.utf8
lv_LV
lv_LV.utf8
lzh_TW
mag_IN
mai_IN
mai_NP
mfe_MU
mg_MG
mg_MG.utf8
mhr_RU
mi_NZ
mi_NZ.utf8
miq_NI
mjw_IN
mk_MK
mk_MK.utf8
ml_IN
mni_IN
mn_MN
mnw_MM
mr_IN
ms_MY
ms_MY.utf8
mt_MT
mt_MT.utf8
my_MM
nan_TW
nan_TW@latin
nb_NO
nb_NO.utf8
nds_DE
nds_NL
ne_NP
nhn_MX
niu_NU
niu_NZ
nl_AW
nl_BE
nl_BE@euro
nl_BE.utf8
nl_NL
nl_NL@euro
nl_NL.utf8
nn_NO
nn_NO.utf8
nr_ZA
nso_ZA
oc_FR
oc_FR.utf8
om_ET
om_KE
om_KE.utf8
or_IN
os_RU
pa_IN
pap_AW
pap_CW
pa_PK
pl_PL
pl_PL.utf8
POSIX
ps_AF
pt_BR
pt_BR.utf8
pt_PT
pt_PT@euro
pt_PT.utf8
quz_PE
raj_IN
rif_MA
ro_RO
ro_RO.utf8
ru_RU
ru_RU.cp1251
ru_RU.koi8r
ru_RU.utf8
ru_UA
ru_UA.utf8
rw_RW
sah_RU
sa_IN
sat_IN
sc_IT
sd_IN
sd_IN@devanagari
se_NO
sgs_LT
shn_MM
shs_CA
sid_ET
si_LK
sk_SK
sk_SK.utf8
sl_SI
sl_SI.utf8
sm_WS
so_DJ
so_DJ.utf8
so_ET
so_KE
so_KE.utf8
so_SO
so_SO.utf8
sq_AL
sq_AL.utf8
sq_MK
sr_ME
sr_RS
sr_RS@latin
ss_ZA
st_ZA
st_ZA.utf8
sv_FI
sv_FI@euro
sv_FI.utf8
sv_SE
sv_SE.iso885915
sv_SE.utf8
sw_KE
sw_TZ
syr
szl_PL
ta_IN
ta_LK
tcy_IN.utf8
te_IN
tg_TJ
tg_TJ.utf8
the_NP
th_TH
th_TH.utf8
ti_ER
ti_ET
tig_ER
tk_TM
tl_PH
tl_PH.utf8
tn_ZA
to_TO
tpi_PG
tr_CY
tr_CY.utf8
tr_TR
tr_TR.utf8
ts_ZA
tt_RU
tt_RU@iqtelif
ug_CN
uk_UA
uk_UA.utf8
unm_US
ur_IN
ur_PK
uz_UZ
uz_UZ@cyrillic
uz_UZ.utf8
ve_ZA
vi_VN
wa_BE
wa_BE@euro
wa_BE.utf8
wae_CH
wal_ET
wo_SN
xh_ZA
xh_ZA.utf8
yi_US
yi_US.utf8
yo_NG
yue_HK
yuw_PG
zh_CN
zh_CN.gb18030
zh_CN.gbk
zh_CN.utf8
zh_HK
zh_HK.utf8
zh_SG
zh_SG.gbk
zh_SG.utf8
zh_TW
zh_TW.euctw
zh_TW.utf8
zu_ZA
zu_ZA.utf8
```

So it seems like the workaround from #27 cannot work because it tries to use `LANG=de_DE.UTF-8` as a workaround but doesn't have this locale installed at all.

eliadevito, 2024-09-05:
flatpak config allow you to configure flatpak apps locale, it is documented here:  https://docs.flatpak.org/en/latest/flatpak-command-reference.html#flatpak-config

you can configure locale that want to use in your flatpak app with
```
flatpak config --set languages "LOCLAE_SEPARATED_BY_SEMICOLON"
```
and after install it with
```
flatpak update
```

for example if you want to install English and German languages you should do
```
flatpak config --set languages "en;de"
flatpak update
```

xarbit, 2024-09-28:
It should still initialize using the fallback locale, which is puzzling since the fallback is hardcoded. 

You can find the relevant code here:  https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/85d844953214bcb24ad82c2c639444f28c0c8a73/entrypoint#L21C61-L21C93

I suspect that the checking mechanism is not functioning correctly, and we may need to consider removing the line: `[ -n "$LANG" ] && export LC_ALL=$LANG || ` and force the hardcoded fallback

xarbit, 2024-09-28:
@CharlemagneLasse  

please keep an eye on this build in: https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/124 
When available, install and let me know if that works for you.

xarbit, 2024-09-28:
> flatpak config allow you to configure flatpak apps locale, it is documented here: https://docs.flatpak.org/en/latest/flatpak-command-reference.html#flatpak-config
>
> you can configure locale that want [...]

That is the right way, but it is a bad experience for regular users. The main reason of the entrypoint script is to catch this problem and allow the application to start nevertheless, even if it is in English. I dont remember when the line `[ -n "$LANG" ] && export LC_ALL=$LANG ||` was introduced, but I do believe it wasn't there initially.

xarbit, 2024-09-28:
That line has always been present, but it doesn’t consistently work in all scenarios when LANG is set in mixed format OS's. .

xarbit, 2024-09-28:
@eliadevito I reworked the locale check, maybe you can review it and run it after it builds? https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/124

xarbit, 2024-09-28:
Build [150301 successful](https://buildbot.flathub.org/#/builders/6/builds/150301)
To test this build, install it from the testing repository:
```
flatpak install --user https://dl.flathub.org/build-repo/133392/com.prusa3d.PrusaSlicer.flatpakref
```

CharlemagneLasse, 2024-09-28:
@xarbit Thank you, it is now finally starting up for me

PR #124 (2024-09-28): fix: #121 + #58
fix: #58 and #121
commit c084d7ae (2024-09-28): fix: #121 + #58 (#124)
commit ff302e13 (2024-09-28): fix: #121 + #58 (#124)
commit 6db4cd73 (2024-12-13): fix: #121 + #58 (#124)
closed 2024-09-28 by xarbit